PaulStoffregen/SD #30: Wildcard in library.properties but filtering architectures

tobozo, 2021-05-12:
Hello and thanks for your awesome library :+1: 

Apparently the master branch of this project has been bundled with Arduino 1.8.14 latest release.
The issue is that the library claims to handle [all architectures](https://github.com/PaulStoffregen/SD/blob/master/library.properties#L9) whereas the code indicates [it clearly does not](https://github.com/PaulStoffregen/SD/blob/master/utility/Sd2PinMap.h#L410).

As a result this library now predates any core-bundled SD.h with a lower version number (such as the one from esp32-arduino), even though that architecture isn't listed in the Sd2PinMap file.

I'm not sure what should happen instead, not erroring to satisfy the wildcard in the properties file, or be exhaustive with the supported architectures, but the only way I could find to solve this error with esp32-arduino was to delete the bundled SD library from the arduino/libraries application folder.

This is a big constraint for me as I'm using your excellent library in other non-esp32 projects and will have to re-import it anyway.

Please let me know what I can do to help, I already suggested to espressif team to raise the version number to a higher value, but this is more a hack than a real fix as the real problem is the wildcard in the properties file.

Thanks again for the great work!

tobozo, 2021-05-14:
followup: https://github.com/espressif/arduino-esp32/issues/5181

per1234, 2021-05-14:
Hi @tobozo.
>Apparently the master branch of this project has been bundled with Arduino 1.8.14 latest release.

The library bundled with the Arduino IDE is this:
https://github.com/arduino-libraries/SD
(as defined [here](https://github.com/arduino/Arduino/blob/1.8.14/build/build.xml#L284-L295))
>this library now predates any core-bundled SD.h with a lower version number (such as the one from esp32-arduino)

The version number is not a factor in library dependency resolution priorities. The reason for the changed behavior is explained here:
https://github.com/arduino/arduino-cli/issues/1292

>This is a big constraint for me

I apologize for the inconvenience. This issue has already been fixed and new releases of both Arduino IDEs and Arduino CLI will be out soon.

tobozo, 2021-05-14:
hey @per1234 thanks for the prompt reply, I'll close this issue since it's already in good hands :+1:

closed 2021-05-14 by tobozo