PaulStoffregen/SD #33: FileImpl not a class or struct name

MichaelBMiner, 2021-09-24:
Hello,

I am using your SD repo as a dependency in my platformio project. When I drill down to the error in SD.h I am told that "FileImpl not a class or struct name"

Where is FileImpl supposed to be defined? How can I fix this?

PaulStoffregen, 2021-09-24:
You need the latest Teensyduino 1.55 core library, which does define FileImpl

PaulStoffregen, 2021-09-24:
Maybe this can help?

https://forum.pjrc.com/threads/68273-Problem-using-Serial1-addMemoryForWrite()?p=289312&viewfull=1#post289312

Different error, but same fundamental problem - PlatformIO using outdated code.

MichaelBMiner, 2021-09-24:
Same issue, unfortunately for me he was able to update his PIO version and it worked. I am using the latest. 

Do I need to run this in vs code? Can I run it from the arduino ide and still use the rtos?

MichaelBMiner, 2021-09-24:
I pointed my code to use the Arduino Teensy 1.55 files. Now I get the two errors

`.pio\libdeps\teensy41\SD\src/SD.h: In member function 'virtual File SDFile::openNextFile(uint8_t)':
.pio\libdeps\teensy41\SD\src/SD.h:126:55: error: no matching function for call to 'File::File(SDFile*)'
  126 |                 if (file) return File(new SDFile(file));`

`.pio\libdeps\teensy41\SD\src/SD.h: In member function 'virtual File SDClass::open(const char*, uint8_t)':
.pio\libdeps\teensy41\SD\src/SD.h:158:55: error: no matching function for call to 'File::File(SDFile*)'
  158 |                 if (file) return File(new SDFile(file));`

PaulStoffregen, 2021-09-24:
Use Arduino.  The Teensyduino installer puts all the right files in place, so this sort of thing will "just work".

MichaelBMiner, 2021-09-24:
I opened the Arduino IDE, I am unable to open my current project. I will look more into it. Thanks for the help